ESP32-Musings/OTA_update_AVR_using_ESP32 #3: Arduino Mega OTA update issue

Prahar08modi, 2022-06-20:
Hello @laukik-hase and @udit7395,

I found this resource very useful as I want to update my Arduino Mega through ESP32. I am glad that the code worked for Arduino Uno by making sure both the modules are powered separately and ensuring a Logic Level Converter between the two. However, I am facing issues interfacing with Arduino Mega. The new code gets written successfully to the Arduino Mega program memory but the readTask returns ESP_FAIL as the verification of the 1st block itself fails.

The screenshot describing the same is attached below.
[image]

I have a basic knowledge of how the code is uploaded through stk500 commands. But, I am not able to understand what exactly is the issue here. Are there any specific requirements or code changes that need to be done before updating Arduino Mega via ESP32?

Thanks in advance:)

udit7395, 2022-06-20:
Hi @Prahar08modi, 

As far as I remember we didn't change anything for Arduino Mega. I would say check if the code uploaded is correct, i.e, upload some different code via Arduino IDE and then try upload `led blink` using OTA to see if the code is uploaded correctly (even though the verification fails). If the code is uploaded correctly via OTA, then the address used for verification needs to be changed. This you will have to look up online either on STK500 docs or the arduino forums.

Prahar08modi, 2022-06-20:
Okay. I'll look for the starting address for Arduino Mega and try to implement the same.

Prahar08modi, 2022-06-21:
Hello @udit7395,

The starting address for flash memory for atmega328 and atmega2560 are same i.e, 0x0000 as per the official datasheet. But I found that the page sizes for atmega328 and atmega2560 are different. Former one has 128 while the latter has 256 bytes. In our code, the hex file parser divides the file into 128 byte blocks. Can this be the issue?

udit7395, 2022-06-21:
@Prahar08modi Could be, You could change the page size and test it. Also maybe it is somehow related to [this issue](https://github.com/ESP32-Musings/OTA_update_AVR_using_ESP32/issues/2) that alambin raised. Try both and let us know the outcome.

Prahar08modi, 2022-06-22:
Hello @udit7395,

I think #2 (The aforementioned issue) has to do with improving the readability of the code as you suggested in the comment itself and does not lead to our desired outcome. I confirm the above statement after making the necessary changes to line numbers 18, 53, and 134 of avr_flash.c. 

As far as changing the page size is concerned, I think it does not change the block sizes as the hex_parser.c does not utilize the BLOCK_SIZE macro. I am not adept at understanding the parsing of hex files so I am unable to make the required changes to hex_parser.c. Can you look at it and make the desired changes for 256 blocks parsing? Or if you can provide a reference to do the same will also be helpful.

Thanks in advance:)

udit7395, 2022-06-23:
@laukik-hase Need your help here. Can you help out @Prahar08modi with his query?

Prahar08modi, 2022-06-27:
Hello @udit7395 and @laukik-hase,

I have been able to solve the issue. My intuition was correct regarding the page size. I understood the hex parser code and did the necessary changes for dividing into 256 bytes blocks. Moreover, I also had to make the following changes to make it work with Atmega2560:

1. The page size in program parameters in function setProgParams,
2. Head array in flashPage and readTask function,
3. loadAddress array in incrementLoadAddress function

I also noticed a bug in the existing code. If the code is exactly divisible by the BLOCK_SIZE, then it does not add 0xFF in the remaining bytes and so the code runs successfully. But, when the hex file is not properly divisible by BLOCK_SIze then there is an error in calculating the block_count. It actually calculates a block less than the desired count due to (idx - 1). This can be solved by changing (idx - 1) to just idx.

Thanks for helping me and leading to the solution @udit7395.

I must admit that this is a great work :+1:

closed 2022-07-05 by udit7395