ESP32-Musings/OTA_update_AVR_using_ESP32 #4: There was no 'Example Configuration' in idf.py menuconfig

Albert245, 2022-11-20:
Hi, I'm now following your project. Found out there was no 'Example Configuration' in the menuconfig. Is there anything changed ? I'm new to this ESP-IDF so please help. 
[image]

mmmahammm, 2022-12-08:
Hi, did struggle with this one as well. After succesful build using ESP-IDF command line tools, idf.py build, it did show up. 

Did change ssid and password using menuconfig, flashed and it worked.

udit7395, 2024-08-23:
[Solution](https://github.com/ESP32-Musings/OTA_update_AVR_using_ESP32/issues/4#issuecomment-1343373568)