ESP32-Musings/OTA_update_AVR_using_ESP32 #7: waitForSerialData timeout

zama737, 2022-12-09:
Noticed time out in waitForSerialData seems not to work as expected. In case MAX_DELAY_MS is intended to be 1000ms, function has 250ms for timer, so timeout is multiplied by 250x. 

```
 int sendBytes(char *bytes, int count)
{
    sendData(TAG_AVR_PRO, bytes, count);
    int length = waitForSerialData(MIN_DELAY_MS, MAX_DELAY_MS);

    if (length > 0)
    {
   
```

```
int waitForSerialData(int dataCount, int timeout)
{
    int timer = 0;
    int length = 0;
    while (timer < timeout)
    {
        uart_get_buffered_data_len(UART_NUM_1, (size_t *)&length);
        if (length >= dataCount)
        {
            return length;
        }
        vTaskDelay(250 / portTICK_PERIOD_MS);
        timer++;
    }
    return 0;
}
```

zama737, 2022-12-11:
It actually makes each page write/read to last 250ms, as data is not available on first loop.

Changed to this, make it flash so much faster!  With timeout=1000, is approximately 1000ms. 
```
#include <rom/ets_sys.h>

int waitForSerialData(int dataCount, int timeout)
{
    int timer = 0;
    int length = 0;

    while (timer < timeout)
    {
        uart_get_buffered_data_len(UART_NUM_2, (size_t *)&length);
        if (length >= dataCount)
        {
            return length;
        }
        ets_delay_us(1000);
        timer++;
    }
    return 0;
}
```

Also note, first parameter is expected count and need to be fixed to the calls as well. 
`int length = waitForSerialData(MIN_DELAY_MS, MAX_DELAY_MS);`
--> 
`int length = waitForSerialData(2, MAX_DELAY_MS);`
`int length = waitForSerialData(BLOCK_SIZE+2, MAX_DELAY_MS);`

udit7395, 2024-08-23:
The delay added was intentional as trade off for higher chances of flashing successfully v/s speed

closed 2024-08-23 by udit7395