ESP32-Musings/OTA_update_AVR_using_ESP32 #10: failed to flash file on arduino uno from esp32s3

prabhatel131, 2024-08-23:
I (11944) : FILE_SERVER (flashFile:35) Writing file to page
I (11984) FILE_SERVER: Found file : Blink.ino.hex (9865 bytes)
I (11984) : FILE_SERVER (flashFile:38) Writing code to AVR memory
I (11984) : avr_pro_mode (resetMCU:78) Starting Reset Procedure
I (12184) : avr_pro_mode (resetMCU:89) Reset Procedure finished
I (12184) : avr_pro_mode (getSync:109) Synchronizing
I (12184) avr_pro_mode: Received 60 bytes: 
I (12184) avr_pro_mode: 0x48 
I (12194) avr_pro_mode: 0x65 
I (12194) avr_pro_mode: 0x6C 
I (12194) avr_pro_mode: 0x6C 
I (12204) avr_pro_mode: 0x6F 
I (12204) avr_pro_mode: 0x20 
I (12204) avr_pro_mode: 0x66 
I (12214) avr_pro_mode: 0x72 
I (12214) avr_pro_mode: 0x6F 
I (12214) avr_pro_mode: 0x6D 
I (12224) avr_pro_mode: 0x20 
I (12224) avr_pro_mode: 0x41 
I (12234) avr_pro_mode: 0x72 
I (12234) avr_pro_mode: 0x64 
I (12234) avr_pro_mode: 0x75 
I (12244) avr_pro_mode: 0x69 
I (12244) avr_pro_mode: 0x6E 
I (12244) avr_pro_mode: 0x6F 
I (12254) avr_pro_mode: 0x21 
I (12254) avr_pro_mode: 0x0A 
I (12254) avr_pro_mode: 0x48 
I (12264) avr_pro_mode: 0x65 
I (12264) avr_pro_mode: 0x6C 
I (12274) avr_pro_mode: 0x6C 
I (12274) avr_pro_mode: 0x6F 
I (12274) avr_pro_mode: 0x20 
I (12284) avr_pro_mode: 0x66 
I (12284) avr_pro_mode: 0x72 
I (12284) avr_pro_mode: 0x6F 
I (12294) avr_pro_mode: 0x6D 
I (12294) avr_pro_mode: 0x20 
I (12294) avr_pro_mode: 0x41 
I (12304) avr_pro_mode: 0x72 
I (12304) avr_pro_mode: 0x64 
I (12314) avr_pro_mode: 0x75 
I (12314) avr_pro_mode: 0x69 
I (12314) avr_pro_mode: 0x6E 
I (12324) avr_pro_mode: 0x6F 
I (12324) avr_pro_mode: 0x21 
I (12324) avr_pro_mode: 0x0A 
I (12334) avr_pro_mode: 0x48 
I (12334) avr_pro_mode: 0x65 
I (12334) avr_pro_mode: 0x6C 
I (12344) avr_pro_mode: 0x6C 
I (12344) avr_pro_mode: 0x6F 
I (12354) avr_pro_mode: 0x20 
I (12354) avr_pro_mode: 0x66 
I (12354) avr_pro_mode: 0x72 
I (12364) avr_pro_mode: 0x6F 
I (12364) avr_pro_mode: 0x6D 
I (12364) avr_pro_mode: 0x20 
I (12374) avr_pro_mode: 0x41 
I (12374) avr_pro_mode: 0x72 
I (12374) avr_pro_mode: 0x64 
I (12384) avr_pro_mode: 0x75 
I (12384) avr_pro_mode: 0x69 
I (12394) avr_pro_mode: 0x6E 
I (12394) avr_pro_mode: 0x6F 
I (12394) avr_pro_mode: 0x21 
I (12404) avr_pro_mode: 0x0A 
I (12404) avr_pro_mode: End of received data
E (12414) avr_pro_mode: Sync Failure: Expected 0x14 0x10 but got 0x48 0x65
I (12414) : avr_pro_mode (setProgParams:116) Setting Prog Parameters
I (12674) avr_pro_mode: Received 2 bytes: 
I (12674) avr_pro_mode: 0x14 
I (12674) avr_pro_mode: 0x10 
I (12674) avr_pro_mode: End of received data
I (12674) avr_pro_mode: Sync Success
I (12684) : avr_pro_mode (setExtProgParams:124) Setting ExProg Parameters
I (12934) avr_pro_mode: Received 20 bytes: 
I (12934) avr_pro_mode: 0x48 
I (12934) avr_pro_mode: 0x65 
I (12934) avr_pro_mode: 0x6C 
I (12934) avr_pro_mode: 0x6C 
I (12934) avr_pro_mode: 0x6F 
I (12944) avr_pro_mode: 0x20 
I (12944) avr_pro_mode: 0x66 
I (12944) avr_pro_mode: 0x72 
I (12954) avr_pro_mode: 0x6F 
I (12954) avr_pro_mode: 0x6D 
I (12964) avr_pro_mode: 0x20 
I (12964) avr_pro_mode: 0x41 
I (12964) avr_pro_mode: 0x72 
I (12974) avr_pro_mode: 0x64 
I (12974) avr_pro_mode: 0x75 
I (12974) avr_pro_mode: 0x69 
I (12984) avr_pro_mode: 0x6E 
I (12984) avr_pro_mode: 0x6F 
I (12984) avr_pro_mode: 0x21 
I (12994) avr_pro_mode: 0x0A 
I (12994) avr_pro_mode: End of received data
E (13004) avr_pro_mode: Sync Failure: Expected 0x14 0x10 but got 0x48 0x65
I (13004) : avr_pro_mode (enterProgMode:130) Entering Pro Mode
I (15014) avr_pro_mode: Received 25 bytes: 
I (15014) avr_pro_mode: 0x52 
I (15014) avr_pro_mode: 0x65 
I (15014) avr_pro_mode: 0x63 
I (15014) avr_pro_mode: 0x65 
I (15014) avr_pro_mode: 0x69 
I (15024) avr_pro_mode: 0x76 
I (15024) avr_pro_mode: 0x65 
I (15024) avr_pro_mode: 0x64 
I (15034) avr_pro_mode: 0x20 
I (15034) avr_pro_mode: 0x66 
I (15044) avr_pro_mode: 0x72 
I (15044) avr_pro_mode: 0x6F 
I (15044) avr_pro_mode: 0x6D 
I (15054) avr_pro_mode: 0x20 
I (15054) avr_pro_mode: 0x45 
I (15054) avr_pro_mode: 0x53 
I (15064) avr_pro_mode: 0x50 
I (15064) avr_pro_mode: 0x33 
I (15064) avr_pro_mode: 0x32 
I (15074) avr_pro_mode: 0x3A 
I (15074) avr_pro_mode: 0x20 
I (15084) avr_pro_mode: 0x50 
I (15084) avr_pro_mode: 0x20 
I (15084) avr_pro_mode: 0x0D 
I (15094) avr_pro_mode: 0x0A 
I (15094) avr_pro_mode: End of received data
E (15104) avr_pro_mode: Sync Failure: Expected 0x14 0x10 but got 0x52 0x65
I (15854) avr_pro_mode: Received 20 bytes: 
I (15854) avr_pro_mode: 0x48 
I (15854) avr_pro_mode: 0x65 
I (15854) avr_pro_mode: 0x6C 
I (15854) avr_pro_mode: 0x6C 
I (15854) avr_pro_mode: 0x6F 
I (15864) avr_pro_mode: 0x20 
I (15864) avr_pro_mode: 0x66 
I (15864) avr_pro_mode: 0x72 
I (15874) avr_pro_mode: 0x6F 
I (15874) avr_pro_mode: 0x6D 
I (15884) avr_pro_mode: 0x20 
I (15884) avr_pro_mode: 0x41 
I (15884) avr_pro_mode: 0x72 
I (15894) avr_pro_mode: 0x64 
I (15894) avr_pro_mode: 0x75 
I (15894) avr_pro_mode: 0x69 
I (15904) avr_pro_mode: 0x6E 
I (15904) avr_pro_mode: 0x6F 
I (15904) avr_pro_mode: 0x21 
I (15914) avr_pro_mode: 0x0A 
I (15914) avr_pro_mode: End of received data
E (15924) avr_pro_mode: Sync Failure: Expected 0x14 0x10 but got 0x48 0x65
E (17944) : avr_flash (flashPage:74) Sync Failure
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x42009aba
0x42009aba: flashFile at /home/externlabs/work/OTA_update_AVR_using_ESP32/file_serving_avr/main/file_server.c:39 (discriminator 1)

file: "./main/file_server.c" line 39
func: flashFile
expression: writeTask(page, block_count)

any suggetion

prabhatel131, 2024-08-23:
@udit7395

udit7395, 2024-08-23:
@prabhatel131 As mentioned in the README, we have only tested on ESP32 dev board (e.g. ESP32-WROVER Kit) or ESP32 core board (e.g. ESP32-DevKitC). Also as you can see in the logs you have shared there is synchronization failure, this difference could be because of the bootloader on Arduino side.

udit7395, 2024-08-23:
Feel free to fork or raise a PR and comment on this issue if you find a fix.

closed 2024-08-23 by udit7395

prabhatel131, 2024-08-23:
The issue was in my arduino side one program was running in it that printing soma data on consol and also sending on uart...when esp requesting for sync resopnse it geeting that data from consol through uart  that you can see in logs ......you priject is working for ESP32S3 wroom -1 N8 dev board.

udit7395, 2024-08-23:
Happy to hear, you have fixed the issue.